alvieboy/ZXInterfaceZ #5: ZX Spectrum 128K compatibility issues

nihirash, 2020-05-24:
As I understand(I might be wrong) - this device uses ports #03, #05, #07 etc.

Main issue of this solution that it makes device totally incompatible with ZX Spectrum 128K and later(+2/+2A/+3/128K compatible computers) - cause IORQULA block only #FE port and doesn't block AY's ports, memory management unit, disk and printer ports(on +3). 

Main issue that original ZX SPECTRUM machines uses bit mask decoding of port instead full address decoding. 

You should use ports that didn't cross with audio ports, memory ports, +3 FDC and internal cetronics port of +2A/+3. 

Information about bit masks present here: https://www.worldofspectrum.org/faq/reference/ports.htm

You may ignore Timex's ports(cause timex hardware is rare now) but 128K spectrum is mainline now.  

Thank you for your work!

I wish you can make it best!

alvieboy, 2020-05-24:
Indeed we may need to choose different port numbers, eventually using full-decoding as well.

Some notes:
 - The IORQULA is only used to mask/update ULA replies for simulating keyboard and audio input. Since (I believe) all IO requests with A0=0 go to a single ULA "register", we are ok here.
 - For other ports, we may also need to use A15-A8 to avoid collision with other spectrums. Note that we don't need conflict checking for interfaces, since none can be connected at same time.

I'll analyse all IO in detail and try to come up with a sensible range for Interface Z, even if it means more complexity on the software side (since we need to manipulate B as well).

Leaving this open while the issue is being analysed.

alvieboy, 2020-05-24:
I added a simple address analysis document. I believe if we use addresses with A0=1, A1=1, A6=1, A7=0 we can avoid clashing with other devices. This gives us 16 IO addresses to use, should be more than enough, and does not require manipulating A8-A15.

commit ee7fee48 (2020-06-11): Move ports around in order to fix #5
closed 2020-06-11 by alvieboy