alvieboy/ZXInterfaceZ #9: roms/videomode.asm missing

rickard-von-essen, 2020-05-27:
Building roms fails with:

```
cd roms; make
make: *** No rule to make target 'videomode.asm', needed by 'INTZ.ROM'.  Stop.
```

alvieboy, 2020-05-27:
Can you confirm all files are there now?

rickard-von-essen, 2020-05-27:
Looks better, but I think it also requires `pasmo` `0.6.0pre` because of the `IFNDEF`. I'll install pasmo from source and report back.

rickard-von-essen, 2020-05-27:
Ok, I could build pasmo from source with the following patch:

```patch
--- pasmo-0.6.0.20070113.0/module.cpp	2006-10-10 23:35:22.000000000 +0200
+++ pasmo-0.6.0.20070113.0-fix/module.cpp	2020-05-27 22:01:41.567789662 +0200
@@ -7,6 +7,7 @@
 #include "trace.h"
 
 
+#include <algorithm>
 #include <fstream>
 #include <stdexcept>
 
@@ -16,6 +17,7 @@
 using std::logic_error;
 using std::pair;
 using std::make_pair;
+using std::for_each;
 
 
 namespace pasmo {
```

And with that I could compile the ROM's. I'll submit  a README for the directory with some notes tomorrow.

closed 2020-05-27 by rickard-von-essen