alvieboy/ZXInterfaceZ #23: Feature: TAP/TZX recording of external tape

alvieboy, 2020-07-28:
Allow recording of external tapes (connected via spectrum EAR) to TAP or TZX.

rickard-von-essen, 2020-07-28:
Would this also work for microdrive programs to, or would it require something more?

alvieboy, 2020-07-28:
> Would this also work for microdrive programs to, or would it require something more?

I never used a microdrive. Do you have information on what actually gets saved in there ? Anyway, I think we can sort out some way to do that, should not be that complex.

alvieboy, 2021-03-22:
More information:
In order to be able to capture tape data, the FPGA should, at every IO read of the ULA port made by the ZX Spectrum, analyse the EAR input and record it. If the value differs from the previous captured EAR, the T-state difference to the previous change and the EAR value shall be stored in a suitable FIFO, in order for the data to be ready by the ESP32 via the regular SPI channel. An interrupt line should notify CPU that we have data to read.
There should be a means to enable/disable capturing of data (i.e. a currenlty unused REG32 bit).
There should be an upper and lower limit on the T-state, but this is reserved for future implementation because it implies adding a not so easy to implement filter.
T-states (high or low) for a given frequency are: 35555555/(2*freq)
Each T-State is ~3.5555Mhz, ~281.25ns.
Upper limit at 20KHz 50% duty cycle is 88 T-states
Lower limit at 60Hz 50% duty cicle is 29629 T-states.

Suggest storing the delta as a 15-bit unsigned value (0 to 32767) and the 16th bit as the EAR value. If no change, then delta shall be all ones.