alvieboy/ZXInterfaceZ #24: Feature: HDMI output

rickard-von-essen, 2020-07-30:
It would be great to have audio/video output via HDMI. It's a more future prof connection than VGA + Analog audio.

rickard-von-essen, 2020-07-30:
Some useful information:

HDMI implementation on a FPGA

https://www.fpga4fun.com/HDMI.html

alvieboy, 2020-07-30:
The FPGA we use does not have SERDES blocks, so the speed is quite limited (should top at 300Mbps), which is not much for HDMI. Note that signals are 10b encoded, so for 300Mpbs you have at max a 30Mhz bit clock. Asides from that, the IO standard is not supported, so it would need to be adapted from LVDS (which is tricky without external ICs - maybe AC coupling works, have to test). 
I took some care in choosing signals on the expansion connector so that there are 4 differential pairs in there. Plan is to do some experiments and see if we can get an accepted/stable output.