alvieboy/ZXInterfaceZ #26: Feature: Kempston joystick interfaces

rickard-von-essen, 2020-07-30:
Support 2x DSUB 9-pin Kempston  (Port 31) joystick interfaces. It is possible to support 2 fire buttons and to have a small amount
of power available on pins 5 and 7 for joysticks that require it (EG: Megadrive pads).

alvieboy, 2020-07-30:
Plan is to have a small Kempston2USB adaptor. I already started work on that. We don't have room currently on the board, those connectors do take quite some space.

Example, with a standard USB connector (the big square one).

[image]

alvieboy, 2020-12-09:
Prototype USB board ordered. 

[image]

alvieboy, 2021-03-23:
Rev 1 of the board is produced and working.

closed 2021-03-23 by alvieboy