alvieboy/ZXInterfaceZ #27: Feature: add diagnostics ROM

rickard-von-essen, 2020-07-30:
Add a diagnostics ROM that can be automatically loaded via a button or by enable it in the web interface etc. To allow the IfZ to be used to diagnose Spectrums with hardware issues.

Retroleum's Diagnostics ROM:

http://blog.retroleum.co.uk/electronics-articles/a-diagnostic-rom-image-for-the-zx-spectrum/

alvieboy, 2020-07-30:
I'm working on the new RAM/ROM layout so we can support multiple roms easily (8k, 16k and 32k).

rickard-von-essen, 2020-07-31:
And some features from the _Multiface_ carts like pokes and

> Multifaces contained an extra 8 kilobytes of RAM that could be used by specially written software. Examples of such software included Genie,[5] a machine code disassembler, and Lifeguard,[5] a program for analysing the memory of a game during play in order to identify which memory locations should be changed to aid cheating.

https://en.wikipedia.org/wiki/Multiface

alvieboy, 2020-08-11:
[image]

Now, leaving the ROM is the hardest part. Will sort it out.
The ROM was loaded via command-line using a ROM file in the SD card.

rickard-von-essen, 2020-08-11:
Great work.

alvieboy, 2020-08-11:
For manual execution, we may need some way to trigger this ROM via the push buttons. Any ideas ? We have two user buttons - one is used for NMI, the other is unused. I'd suggest a long keypress (like 2secs) would trigger execution of a diagnostics ROM.

rickard-von-essen, 2020-08-11:
Yes, it would be great if the diag rom was stored on the flash chip. Long press USR? That should be fine, would that in practice mean long press USR and then hit reset?

Tailzer, 2021-03-15:
Could you detect usr button held at power on like the Dandanator does?

rickard-von-essen, 2021-03-15:
> Could you detect usr button held at power on like the Dandanator does?

Currently it's implemted by holding IOB more than 2 seconds. Is there any reason USR at power on/reset would be better? :thinking_face:

rickard-von-essen, 2021-03-15:
Closing since this is already implemented.

closed 2021-03-15 by rickard-von-essen

Tailzer, 2021-03-15:
No only just frees the long press for other functions in the future,
plugging in for diagnostics is something that would nearly always be
associated with power on?
Great project, watching with interest!

alvieboy, 2021-03-16:
The idea is that this could be configurable in the future. In fact, plan is to be able to confugure short/long presses on the board buttons and support long presses on joystick/keypad buttons/USB keyboards as well.  All these should map to a user-configurable action.
@Tailzer I could in theory add such a press-on-reset behaviour for one of the buttons, but I want to reserve the press-on-reset on at least one of them for the purpose of "factory reset".