alvieboy/ZXInterfaceZ #31: esp32: make flash fails with ./filesize: Command not found

rickard-von-essen, 2020-08-14:
```
$ make flash                                                                                                              
fatal: No names found, cannot describe anything.
Toolchain path: /home/rickard/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r2
Compiler version: 8.2.0
Python requirements from /home/rickard/src/esp-idf/requirements.txt are satisfied.

App "esp32_interfacez" version: d416435-dirty
GEN fpga.bin
make: ./filesize: Command not found
make: *** [Makefile:16: fpga.bin] Error 127
```
I can't find any ref other ref to `filesize`.

alvieboy, 2020-08-14:
The tool was missing. Please check if last commit fixed it.

rickard-von-essen, 2020-08-14:
Works now!

closed 2020-08-14 by rickard-von-essen