alvieboy/ZXInterfaceZ #56: /req/wifi doesn't report assigned DHCP IP

rickard-von-essen, 2020-11-28:
Currenlty the API `/req/wifi` returns

```json
{
  "mode" : "ap", 
  "sta": {
   	"ssid": "ssid",
        "hostname": "interfacez", 
        "ip": "dhcp", 
        "netmask": "255.255.255.0", 
        "gw": "127.0.0.1", 
  },
  "ap" : {
   	"ssid": "ssid",
        "hostname": "interfacez",
        "channel": 1,
        "ip" : "127.0.0.1",
        "netmask" : "255.255.255.0",
        "gw" : "127.0.0.1"
  }
}
```

Where `ip: dhcp` but I think it would be better to add `"dhcp": true, "ip": "<currently leased IP>"`.

alvieboy, 2020-11-29:
Definitely, it should return the IP address in there. I'll change it as per your proposal

alvieboy, 2020-11-30:
Ok now I undestand why it looks like this. This relates to the WiFi configuration, not to the WiFi status. For status, I propose adding a new node which would  the current IP configuration. Looks OK for you?

rickard-von-essen, 2020-12-02:
That's fine. I could also include the MAC etc. Also there is no way of setting the config from the API either.

alvieboy, 2020-12-22:
How do you prefer to handle the sets for WiFI? JSON or POST parameters?

rickard-von-essen, 2020-12-22:
POST with json body.

alvieboy, 2020-12-22:
For "scan" operations, via /req/scan, the operation will return "success":"false",  and error field will not be present while the scan is in progress. You can use the scan results to configure WiFi after you retrieve the SSID from the scan.

Does this make sense for the JSON part:

WiFi Disabled:
	"mode": "disabled"
        
        
WiFi as access point (dhcp server implicit):
	"mode": "ap",
        "ip": "x.x.x.x",
        "netmask": "x.x.x.x",
        "ssid": "theapssid",
        "password": "theappassword".
        "channel": "8"
        

WiFi station (dhcp):
	"mode": "sta",
        "inet": "dhcp",
        "ssid": "theapssid",
        "password": "theappassword_or_empty"

WiFi station (static IP):
	"mode": "sta",
        "inet": "static",
        "ip": "x.x.x.x",
        "netmask": "x.x.x.x",
        "gw": "x.x.x.x",
        "ssid": "theapssid",
        "password": "theappassword_or_empty"