alvieboy/ZXInterfaceZ #68: Support saving screenshots to SD card

rickard-von-essen, 2021-03-23:
Support grabbing the screen content, either with the USR menu (omitting the menu) or using a custom key on a USB connected device. Save the screen to a predefined path either on the SD Card or a USB mass storage device.

rickard-von-essen, 2021-03-23:
i'm guessing there is no realtime clock so we probably need to keep a counter and use it in the filename. (`/sdcard/screenshoot/interfacez-scr-0001.png`)

alvieboy, 2021-03-23:
This is intimately linked to the quick snapshot feature. 
My idea for the file naming is to in a first instance (i.e., first activation after a ZX Spectrum reset [not Interface Z reset]) to present a file chooser dialog where you can select the directory and the file name template (which will be "preloaded" if you have since loaded a tape, but you can edit it freely). Then, auto increment counters will be used for the file name (incrementing until there is a free "slot" for the file, so you don't overwrite old snapshots. The template itself can either be hardcoded (selectedname-XXXX.ext) or manually input in the future.